florasim/flora #1: Adding NED files

Tanganelli, 2018-03-09:
Hi,
I am relatively new to omnetpp, however, I am trying to extend your simulator to add different kinds of lora nodes and different kinds of MAC layers. I followed the information on the webpage and I added the flora as an existing project to omnet, the example works perfectly. 

Then I tried different options:
1- I directly created a new node and a new MAC in the flora project, but when I compile from the IDE, nothing is added to the generated Makefile.

2- I created another project that depends on flora, but when I try to run a new network with my new node deployed in the new project I have to select an executable. If I choose my project executable I'll get an error : 
"Class "inet::physicallayer::LoRaMedium" not found -- perhaps its code was not linked in, or the class wasn't registered with Register_Class(), or in the case of modules and channels, with Define_Module()/Define_Channel() -- in module (omnetpp::cModule) LoRaNetworkTest (id=1), during network setup"
This is also raised if I change the executable of flora to opp_run.

In general, I don't understand how omnetpp should update your Makefiles when new classes are added.

Sorry for the long question.
Giacomo

mariuszslabicki, 2018-03-13:
Hi,

Just to be sure - what about test scenario, if you run the provided test files do they run correctly?

I remember that issues with creating classes were not obvious. I found it a little tricky to define it correctly in ned files and cpp sources. From the details that you provided it is not easy to find out what is not working. Have you checked the OMNeT documentation? The description for creating new classes is described in 7.12.

Regarding makefiles - omnet provides a tool called "opp_makemake" which can create makefiles for you. Have you seen that? The details are also in documentation - 9.2 Using opp_makemake and Makefiles. The syntax is also complex, and sometimes it is easier for tests to change Makefile by hand.

You can start from above steps, then let me know if you face further problems.

adeel1710, 2020-02-03:
Hi. I am trying to run the FLoRa simulator as per the instructions using Omnet++5.2.1 and INET Framework 3.6.3 . When i try to build the project, I get an error that says this:  

make[1]: Entering directory '/c/Users/AhmedAde/Downloads/flora-master (1)/flora-master/src'
MSGC: LoRa/LoRaMacControlInfo.msg
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `opp_msgc -s _m.cc -I. -IC:/Users/AhmedAde/Downloads/inet-3.6.3-src (1)/inet-3.6.3-src (1)/inet/src LoRa/LoRaMacControlInfo.msg'
make[1]: *** [Makefile:156: LoRa/LoRaMacControlInfo_m.h] Error 1
make[1]: Leaving directory '/c/Users/AhmedAde/Downloads/flora-master (1)/flora-master/src'
make: *** [Makefile:2: all] Error 2

[image]

Is this an issue with the make file because it is hard for me to understand where this syntax error is . It would be great if you could help me out here.

mariuszslabicki, 2020-02-03:
Are you building it on Windows?

adeel1710, 2020-02-03:
Hi yes. I am building it on windows 10

mariuszslabicki, 2020-02-03:
I am not able to verify it as I don't have an access to any machine with Win 10.

adeel1710, 2020-02-03:
Oh. Could you let me know what could be the issue with this ? Is it with the make file ?

I will also try to execute it with Linux later , but as of now I have windows OS.

gpremsan, 2020-12-23:
@adeel1710 The issue with the Makefile was fixed previously. It would be best to run the software on Ubuntu, even using VirtualBox if necessary. If you still face problems with building FLoRa on Ubuntu, please open a new issue.

closed 2020-12-23 by gpremsan