florasim/flora #2: Compile error: Definition of dllimport static field not allowed

shawplee, 2018-04-02:
Hi,
I am new to omnetpp and LoRa.  Trying to run your LoRa simulation but getting error: Definition of dllimport static field not allowed.

I am using Omnet++ 3.5.1 with INET 3.6.4.  Platform is Windows.

The output of console:
...
In file included from src/LoRa/LoRaRadio.cc:20:
../inet/src\inet/common/ModuleAccess.h:98:4: warning: 'inet::getModuleFromPar' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
T *getModuleFromPar(cPar& par, const cModule *from, bool required)
   ^
../inet/src\inet/common/ModuleAccess.h:95:13: note: previous declaration is here
INET_API T *getModuleFromPar(cPar& par, const cModule *from, bool required = true);
            ^
../inet/src\inet/common/ModuleAccess.h:95:1: note: previous attribute is here
INET_API T *getModuleFromPar(cPar& par, const cModule *from, bool required = true);
^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/smartlab/Desktop/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:31:24: error: definition of dllimport static field not allowed
simsignal_t LoRaRadio::minSNIRSignal = cComponent::registerSignal("minSNIR");
                       ^
src/LoRa/LoRaRadio.h:31:7: note: attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/smartlab/Desktop/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:32:24: error: definition of dllimport static field not allowed
...

The errors specifically are referring to the following line in LoRaRadio.cc:

simsignal_t LoRaRadio::minSNIRSignal = cComponent::registerSignal("minSNIR");
simsignal_t LoRaRadio::packetErrorRateSignal = cComponent::registerSignal("packetErrorRate");
simsignal_t LoRaRadio::bitErrorRateSignal = cComponent::registerSignal("bitErrorRate");
simsignal_t LoRaRadio::symbolErrorRateSignal = cComponent::registerSignal("symbolErrorRate");
simsignal_t LoRaRadio::droppedPacket = cComponent::registerSignal("droppedPacket");

How to resolve the problem?  Thank you

regards,

Shaw

mariuszslabicki, 2018-04-02:
Hi,

Thanks for the info. We try to solve the problem - I don't have access to a machine with Windows, so debugging is impossible for me right now. I will put my hands on the Win machine in next few days and try to solve it. I hope we will be able to patch it. I will let you know in this topic when we solve it.

AndreyBiller, 2018-04-08:
Good evening,

[OMNeT++ 5.2.1 ; INET framework 3.6.3]

The same problems on Windows machine, but also trying on Linux and get next one message
[image]

Thank you, best regards,

Andrey

gpremsan, 2018-04-10:
Hi Andrey, the issue you have reported should now be fixed with commit 468108a on master. There was an issue with the Makefile. Please pull the latest version of the repository to get the changes.

mariuszslabicki, 2018-04-12:
Hi

Regarding the problem with compiling FLoRa in Windows. We still haven't solved the problem how to compile FLoRa with clang. But there is a workaround how to use the software:

When you change the flag PREFER_CLANG in configure.user to 'no'. In that case the gcc will be used to compile project. I have just checked it, and FLoRa runs correctly.

More information about changing this flag you can find in Install Guild, section Build Options.
https://omnetpp.org/doc/omnetpp/InstallGuide.pdf

closed 2018-04-13 by mariuszslabicki