florasim/flora #7: building project problem

gnob417, 2018-05-11:
Hi, I wanna build Flora, however after doing "build project", error with below log occurred.
what is the problem?

```
src/LoRa/LoRaRadio.h:156:10: note: previous declaration is here
    void startRadioModeSwitch(RadioMode newRadioMode, simtime_t switchingTime);
         ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:153:17: warning: 'inet::physicallayer::LoRaRadio::completeRadioModeSwitch' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
void LoRaRadio::completeRadioModeSwitch(RadioMode newRadioMode)
                ^
src/LoRa/LoRaRadio.h:157:10: note: previous declaration is here
    void completeRadioModeSwitch(RadioMode newRadioMode);
         ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:166:33: warning: 'inet::physicallayer::LoRaRadio::getTransmissionInProgress' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
const ITransmission *LoRaRadio::getTransmissionInProgress() const
                                ^
src/LoRa/LoRaRadio.h:226:34: note: previous declaration is here
    virtual const ITransmission *getTransmissionInProgress() const override;
                                 ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:174:33: warning: 'inet::physicallayer::LoRaRadio::getReceptionInProgress' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
const ITransmission *LoRaRadio::getReceptionInProgress() const
                                ^
src/LoRa/LoRaRadio.h:227:34: note: previous declaration is here
    virtual const ITransmission *getReceptionInProgress() const override;
                                 ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:182:37: warning: 'inet::physicallayer::LoRaRadio::getTransmittedSignalPart' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
IRadioSignal::SignalPart LoRaRadio::getTransmittedSignalPart() const
                                    ^
src/LoRa/LoRaRadio.h:229:38: note: previous declaration is here
    virtual IRadioSignal::SignalPart getTransmittedSignalPart() const override;
                                     ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
[... 496 lines cut ...]
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
src/LoRa/LoRaRadio.cc:612:17: warning: 'inet::physicallayer::LoRaRadio::setCurrentTxPower' redeclared without 'dllimport' attribute: previous 'dllimport' ignored [-Winconsistent-dllimport]
void LoRaRadio::setCurrentTxPower(double txPower)
                ^
src/LoRa/LoRaRadio.h:205:10: note: previous declaration is here
    void setCurrentTxPower(double txPower);
         ^
src/LoRa/LoRaRadio.h:31:7: note: previous attribute is here
class INET_API LoRaRadio : public FlatRadioBase //: public PhysicalLayerBase, public virtual IRadio
      ^
../inet/src\inet/common/INETDefs.h:52:23: note: expanded from macro 'INET_API'
#  define INET_API    OPP_DLLIMPORT
                      ^
C:/Users/Youngjune/Desktop/FloRa/omnet/omnetpp-5.2.1-src-windows/omnetpp-5.2.1/include/omnetpp/platdep/platdefs.h:33:37: note: expanded from macro 'OPP_DLLIMPORT'
#  define OPP_DLLIMPORT  __declspec(dllimport)
                                    ^
52 warnings and 5 errors generated.
make: *** [Makefile:155: out/clang-debug//src/LoRa/LoRaRadio.o] Error 1

17:42:47 Build Finished (took 7s.933ms)
```

mariuszslabicki, 2018-05-11:
Could you provide more information what is your environment? Do you run it on Windows?

gnob417, 2018-05-11:
yes, i run it on windows 10.
OmNet and INET version are same as your description said.

i tried again but, below error occurred.

```
20:52:49 **** Build of configuration gcc-debug for project flora ****
make MODE=debug all 
src/LoRa/LoRaGWMac.cc
In file included from src/LoRa/LoRaGWMac.cc:16:0:
src/LoRa/LoRaGWMac.h:19:10: fatal error: inet/physicallayer/contract/packetlevel/IRadio.h: No such file or directory
 #include "inet/physicallayer/contract/packetlevel/IRadio.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:155: out/gcc-debug//src/LoRa/LoRaGWMac.o] Error 1

20:52:49 Build Finished (took 408ms)
```

mariuszslabicki, 2018-05-12:
Right now, we face some problems with compatibility with clang. But the tool can be compiled with gcc. Both toolchains are provided with omnet, so you have to just change default compiler in the configuration file and rebuild environment again.

You can change it in the configure.user file, there is a line:
PREFER_CLANG=yes

If you change it to no, and build the environment again (compile omnet + inet), it will works.

Please let me know if you solve it.

gnob417, 2018-05-13:
I did after modifying it to  PREFFR_CLANG=no, however same error above occurs again.
inet is compiled with no error.
only FloRa has compiling issue.

what is this part of error message from above?
```
src/LoRa/LoRaGWMac.h:19:10: fatal error: inet/physicallayer/contract/packetlevel/IRadio.h: No such file or directory
 #include "inet/physicallayer/contract/packetlevel/IRadio.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

mariuszslabicki, 2018-05-13:
Ok, so probably you are facing problems with correct directory structures - we are solving both problems right now. Could you try source from this branch?: https://github.com/mariuszslabicki/flora/tree/testmakefile

In the main version there is a problem with executable name. And this problem is solved in this branch, but it is not moved to the main branch. 

The problem with clang (as I mentioned above) can be solved with changing to gcc. If you did that that is fine, just please make sure that you recompiled omnet and inet after this change - as I remember, you have to recompile the whole environment.

Feel free to ask further.

mariuszslabicki, 2018-05-15:
All sources for FLoRa, including sources for ADR mechanism are included in git repo. Check "src" directory.

Carlos-Gonzaga93, 2020-01-29:
I have an error when compiling flora could you help me, I am using windows 10
n file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:19:10: fatal error: 'inet/physicallayer/contract/packetlevel/IRadio.h' file not found
#include "inet/physicallayer/contract/packetlevel/IRadio.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [Makefile:146: ../out/clang-release/src/LoRa/LoRaGWMac.o] Error 1
make[1]: Leaving directory '/c/Users/Usuario/Documents/Programas/OMNET++/omnetpp-5.6/samples/flora/src'
make: *** [Makefile:2: all] Error 2
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

21:24:06 Build Failed. 3 errors, 0 warnings. (took 2s.899ms)

mariuszslabicki, 2020-01-29:
Which operating system are you using?

Carlos-Gonzaga93, 2020-01-29:
windows 10

mariuszslabicki, 2020-01-29:
Well, I cannot help with that as I don't have an access to a machine with it. In the past we have some issues with Windows. I can just recommend to check if software versions are right, and maybe previous posts in this thread can suggest a solution.

Carlos-Gonzaga93, 2020-01-29:
In which operating system it is recommended to install it

mariuszslabicki, 2020-01-29:
We have tested it with Ubuntu. I was also using it in mac os x.

mariuszslabicki, 2020-01-29:
I think that "testmakefile" was an old test branch that then we have pulled to the main branch.

Carlos-Gonzaga93, 2020-01-29:
I have tried the ubnet version of omnet ++ 5.2.1 on ubuntu and it gives me an error when trying to download inet as follows An internal error occurred during: "Download project descriptors".
Invalid thread access

mariuszslabicki, 2020-01-29:
You can download inet from the inet webpage and the add it to omnet.

Carlos-Gonzaga93, 2020-01-29:
tambien lo intente de esa forma y tengo error

mariuszslabicki, 2020-01-29:
Sorry, I am not supporting omnet and inet in general.

closed 2020-01-29 by mariuszslabicki

Carlos-Gonzaga93, 2020-01-30:
497/5000
I already installed in ubuntu and I get an error when executing the LoRaNetworkTest.ini example
The error shows that the file Flora.exe cannot find any help?
Exception occurred executing command line.
Cannot run program "C: / Users / User / Documents / Programs / OMNET ++ / omnetpp-5.6 / samples / flora / src / flora.exe" (in directory "C: \ Users \ User \ Documents \ Programs \ OMNET ++ \ omnetpp- 5.6 \ samples \ flora \ simulations "): CreateProcess error = 2, The system cannot find the specified file

mariuszslabicki, 2020-01-30:
Yes. The system cannot find the specified file.

Carlos-Gonzaga93, 2020-01-30:
Mariuszslabicki / Is there any way to fix the error

ierms, 2020-04-26:
Hello, I am trying to start a simulation project with Flora. I have instaalled Omnet 5.4.1 which comes with INET and the flora project in the samples directory. While trying to compile the first example I get the following error. Would you please help me?
09:21:52 **** Incremental Build of configuration gcc-release for project flora ****
make MODE=release all 
make[1]: Entering directory '/home/rodrigo/omnetpp-5.4.1/samples/flora/src'
LoRa/LoRaGWMac.cc
In file included from LoRa/LoRaGWMac.cc:16:0:
LoRa/LoRaGWMac.h:19:10: fatal error: inet/physicallayer/contract/packetlevel/IRadio.h: No such file or directory
Makefile:144: recipe for target '../out/gcc-release/src/LoRa/LoRaGWMac.o' failed
 #include "inet/physicallayer/contract/packetlevel/IRadio.h"
make[1]: Leaving directory '/home/rodrigo/omnetpp-5.4.1/samples/flora/src'
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Makefile:2: recipe for target 'all' failed
compilation terminated.
make[1]: *** [../out/gcc-release/src/LoRa/LoRaGWMac.o] Error 1
make: *** [all] Error 2

09:21:52 Build Finished (took 240ms)

rogerpueyo, 2020-04-26:
Hi @ierms,

I've successfully run FLoRa with OMNet++ 5.2.1 and INET 3.6.4, with Ubuntu and Debian. Give it a try using these specific versions.

Cheers!

ierms, 2020-04-26:
Es que no logro instalar esas versiones... Da error en el make

rogerpueyo, 2020-04-26:
May I ask you what is the specific error you get?

hxyzha, 2020-12-04:
Hello, 
I installed omnetpp5.6.3+inet 4.2.1 + Flora in windows 10.
I don't have problem with Inet I have a problem with Flora. It gives me many errors such as::

'analogModelType': no such parameter	LoRaMedium.ned	/flora/src/LoRaPhy	
'antennaType': no such parameter	LoRaGWRadio.ned	/flora/src/LoRa	
'antennaType': no such parameter	LoRaRadio.ned	/flora/src/LoRa	
'carrierFrequency': no such parameter	LoRaGWRadio.ned	/flora/src/LoRa	
'carrierFrequency': no such parameter	LoRaRadio.ned	/flora/src/LoRa	
'IPv4NetworkConfigurator': no such module type	package.ned	/flora/simulations	
'IUDP': no such module interface type	LoRaGW.ned	/flora/src/LoraNode	
'IWiredNic': no such module interface type	LoRaGW.ned	/flora/src/LoraNode	
'pathLossType': no such parameter	LoRaMedium.ned	/flora/src/LoRaPhy	
'propagationType': no such parameter	LoRaMedium.ned	/flora/src/LoRaPhy	
'receiverType': no such parameter	LoRaGWRadio.ned	/flora/src/LoRa	
'receiverType': no such parameter	LoRaRadio.ned	/flora/src/LoRa	
'transmitterType': no such parameter	LoRaGWRadio.ned	/flora/src/LoRa	
'transmitterType': no such parameter	LoRaRadio.ned	/flora/src/LoRa	
a simple module cannot extend a compound module ('CsmaCaMac')	LoRaMac.ned	/flora/src/LoRa	
cannot assign parameters of a submodule of unknown type	package.ned	/flora/simulations	

How can resolve this problem please?

rogerpueyo, 2020-12-04:
> How can resolve this problem please?

Please check these comments on another issue:
 - https://github.com/mariuszslabicki/flora/issues/28#issuecomment-727225258
 - https://github.com/mariuszslabicki/flora/issues/28#issuecomment-727231079

Those are OS + OMNeT + INET + FLoRa known to work.

danzuha, 2020-12-05:
Hi there,

I think its better you follow following step in order to get flora

1. Download virtual box
2. Install ubuntu 16.02lts
3. Install omnett 5.2.1
4. Install inet 3.6.3
5. And build flora

hxyzha, 2020-12-05:
I have already ubuntu 20.04 installed but I cannot install INET and Flora. It generates errors when I build project.
On windows 10 I installed omnetpp5.2.1 but inet 3.6.3 and Flora also doesn't work.

What I must do please?

danzuha, 2020-12-05:
> I have already ubuntu 20.04 installed but I cannot install INET and Flora. It generates errors when I build project.
> 
> On windows 10 I installed omnetpp5.2.1 but inet 3.6.3 and Flora also doesn't work.
> 
> 
> 
> What I must do please?
> 
> 
> 
> 

It will not work on ubuntu 20.04 nor windows 10 as there will be so many downgrade need to be done with the java etc. 

Use ubuntu 16.04LTS, its seamless.

Elanktar, 2021-07-07:
Hello everyone! 

I'm trying to build the project as well. I'm using INET 4.3.2 and Omnet++ 6.0pre11.
The flora project is already referencing to INET and I also tried to put the line "PREFER_CLANG=no" in the config.user file.

I'm still meeting this problem and I'm running out of ideas:

make MODE=debug all 
make[1]: Entering directory '/c/Users/bidule/Downloads/omnetpp-6.0pre11/samples/flora/src'
Creating shared library: ../out/clang-debug/src/libflora_dbg.dll
lld-link: error: duplicate symbol: inet::SharingTagSet::setTag(int, inet::IntrusivePtr<inet::TagBase const> const&)
>>> defined at ../../inet4.3/src\inet/common/packet/tag\SharingTagSet.h:163
>>>            ../out/clang-debug/src/LoRa/LoRaGWMac.o
>>> defined at libINET_dbg.dll

lld-link: error: duplicate symbol: inet::SharingRegionTagSet::operator=(inet::SharingRegionTagSet const&)
>>> defined at ../../inet4.3/src\inet/common/packet/tag\SharingRegionTagSet.h:283
>>>            ../out/clang-debug/src/LoRa/LoRaMacFrame_m.o
>>> defined at libINET_dbg.dll

lld-link: error: duplicate symbol: inet::OperationalMixin<omnetpp::cSimpleModule>::numInitStages() const
>>> defined at ../../inet4.3/src\inet/common/lifecycle\OperationalMixin.h:51
>>>            ../out/clang-debug/src/LoRa/LoRaGWMac.o
>>> defined at libINET_dbg.dll

lld-link: error: duplicate symbol: inet::OperationalMixin<omnetpp::cSimpleModule>::isUp() const
>>> defined at ../../inet4.3/src\inet/common/lifecycle\OperationalMixin.h:76
>>>            ../out/clang-debug/src/LoRa/LoRaGWMac.o
>>> defined at libINET_dbg.dll

lld-link: error: duplicate symbol: inet::OperationalMixin<omnetpp::cSimpleModule>::isDown() const
>>> defined at ../../inet4.3/src\inet/common/lifecycle\OperationalMixin.h:77
>>>            ../out/clang-debug/src/LoRa/LoRaGWMac.o
>>> defined at libINET_dbg.dll
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:144: ../out/clang-debug/src/libflora_dbg.dll] Error 1
make[1]: Leaving directory '/c/Users/bidule/Downloads/omnetpp-6.0pre11/samples/flora/src'
make: *** [Makefile:5: all] Error 2
"make MODE=debug all" terminated with exit code 2. Build might be incomplete.

If someone has an idea I would be really greatful :)