florasim/flora #8: receiver implementation

gnob417, 2018-05-23:
Hi, did you implement two receive window? 
when I run a simulation, all nodes seems to detect packets from other nodes, however, nodes only listen after transmitting a packet according to LoRaWAN specification.

and they can not listen uplinks among nodes because they can listen only I/Q inverted signal which downlink uses.

these nodes are not likely to use Class A device, am i right?

mariuszslabicki, 2018-05-27:
Hi

We have implemented two receiving windows. In a file LoRaMac.cc you can find implemented a Finite State Machine, which has (among others) two states: Receiving_1, and Receiving_2. They are corresponding with two receiving windows in LoRa.

I don't understand the part with uplinks and I/Q inverted signal, what do you mean by that?

Yes, there are only Class A devices.

gnob417, 2018-06-12:
as far as i know, nodes could not listen other nodes' uplinks, however, GUI interface seems that they could.
and I can not see any downward packets transmitted from gateway. 
is it implemented in code only, not GUI? (If gateway send a downlink, it can not receive any uplink for a while because lorawan gateway is not full-duplex)

mariuszslabicki, 2018-06-12:
I think that it is impossible to have different behaviour in code and in GUI, as the code defines how the simulation behaves. I don't understand the issue with uplinks which are received by other nodes. Please explain more details.

ChArendt92, 2018-07-04:
Hi,

@gnob417 over the air packets in omnet are displayed as if they are being sent to all wireless nodes in the network, as there is no dedicated connection between them. The radio receiver module of the nodes then has to decide if it is something it needs to process or not. Typically, the uplink packets should then only be processed in the Gateways

@mariuszslabicki I have two additional questions regarding the receiver windows. In the LoRaMac of the Nodes I can see the FSM containing the two receiving windows on which the nodes are listening for downlink messages. However, the gateway does not seem to consider these, is that correct? So the Gateway only waits for its duty cycle, but if it did not send for a while it just sends out it's message without considering the receiving windows. Am I missing something here?
Additionally, is there another downlink frequency defined for receiving window 2 in flora?

Thanks and Regards

Christian

gnob417, 2018-08-14:
@mariuszslabicki hi again! 
in the real test-bed, a server forwards downlinks to a gateway with the time when the gateway should transmit them. because end node waits for downlink only after uplinks (for 2~3 second), it may not receive downlinks if the gateway does not transmit a packet on the right time. 
Is this feature implemented in FLoRa?

And, downlinks from the gateway seem invisible on GUI.

mariuszslabicki, 2018-08-22:
Hi @gnob417 
In our case, the GW node doesn't have any logic to control time. All decisions and delays are made in NetworkServer App. So in the current implementation when NetworkServer wants to send back an ADR message it creates the message and scheduled it after 1.2 sec - unfortunately, this is a magic number, which we have tuned. This is the exact time which we need to fulfil receiving window requirements. If you want to improve it, you should add logic in GW where you use expected time of transmission, and this time should be written in a packet which is sent from NetworkServer to GW. As we wanted to omit this logic, in our case we simplified this.

Regarding the visibility of downlinks - strange behaviours, as I remember it was working. We are not using GUI too much, but I am sure that transmission is successful from GW to end node. Are you sure that packet is transmitted?

spiroskat, 2019-07-30:
Hi,
i would like to ask, if i want to implement a logic in a high level, for example in the server i want to send a downlink message in various moments etc and i want to use FLoRa as a black box, i don't care how lora actually works in lower levels, i only want to have a logic running in a lora network, where is recommended to write this code , and is there something like an API (of the FLoRa) that i can use such as for example send_to_lora_nodes(message) ?

mariuszslabicki, 2019-08-02:
If you want to be able to send a message from server, the best idea is to add it in the NetworkServer component.

spiroskatsampiris, 2019-11-15:
Hi, i have noticed every time i run the example, the energy consumption is the same value. Why this happens? Also, i would like to ask if i change the mobility of the nodes, the results will remain the same for the energy and power consumption? If yes, why ? Could you give me a reference of your energy model?

mariuszslabicki, 2019-11-15:
I don't understand the question. Please give more description. We don't consider any mobility model, nodes are static. You can find description in source file for energy consumption model.

spiroskatsampiris, 2019-11-15:
Every time a execute the example, loraNetworkTest, i notice that in the results, every node has the exact same power consumption statistics. Specifically, if i run the example with one node, the values of totalEnergyConsumed (scalar) and powerEnergyConsumption are the same, every time i run it.

mariuszslabicki, 2020-01-24:
For how long you run the simulation? And are you sure that you change the random seed? If you run the same simulation twice for the same random init, the results are always the same.

h20180118, 2020-03-18:
Hi, lora gateway can support multiple reception paths. Does flora implement this feature?

mariuszslabicki, 2020-03-19:
Do you mean multipath propagation? I don't understand what you are asking for.

h20180118, 2020-03-19:
I understand lora gateway can receive from 8 different channels simultaeneously (as per semtech), does flora include this feature?

mariuszslabicki, 2020-03-19:
No. FLoRa uses just one channel. It can receive 6 different SF on parallel.

h20180118, 2020-03-23:
Thank you for the response. Also in the code gateway's carrier frequency is 2450Mhz and lora node's carrier frequency is 868Mhz. How 2 different frequencies are able to communicate with each other?
And in pass loss model, the maximum sensitivity is set to -137dbm. In order to compute  range, the maximum sensitivity is supposed to change based on selected SF right?

mariuszslabicki, 2020-03-23:
Could you please link to particular files? I don't know which module you mean.

h20180118, 2020-03-23:
In LoRaRadio.ned,  carrierFrequency = 868 MHz;
In LoRaGWRadio.ned, carrierFrequency = 2450 MHz;
These two frequencies are different.
And in module LoRaLogNormalShadowing.cc, 
The function given below has fixed maximum sensitivity. The computed range should change based on SF right?
m LoRaLogNormalShadowing::computeRange(W transmissionPower) const
{
    // parameters taken from paper "Do LoRa Low-Power Wide-Area Networks Scale?"
    double PL_d0_db = 127.41;
    double max_sensitivity = -137;
    double trans_power_db = round(10 * log10(transmissionPower.get()*1000));
    EV << "LoRaLogNormalShadowing transmissionPower in W = " << transmissionPower << " in dBm = " << trans_power_db << endl;
    double rhs = (trans_power_db - PL_d0_db - max_sensitivity)/(10 * gamma);
    double distance = d0.get() * pow(10, rhs);
    return m(distance);
}
Sorry, I'm not able to attach the files. I have mentioned the module names and function above.

mariuszslabicki, 2020-03-23:
As far as I remember, those carrier frequencies are not used in code, we just have them there due to the inheritance from upper classes.

Regarding the sensitivity, this is the minimal sensitivity of LoRa chip. It is used to evaluate maximal possible distance as I remember.  Sensitivity is evaluated there:
https://github.com/mariuszslabicki/flora/blob/master/src/LoRaPhy/LoRaReceiver.cc#L253

mariuszslabicki, 2021-04-10:
No more posts, let me close it.

closed 2021-04-10 by mariuszslabicki