florasim/flora #10: Port to Inet 4

aarizaq, 2019-03-21:
Sorry for the nuisances. I am trying to port the Flora code to inet/inetmanet 4.x and omnet 5.4.1 and more or less I think that I have a first version. There are several thinks that I want to actualize like the error model, but this is not the thing that I want to request.

If you are not too busy, could you check the port? You know how the software should work and I you can check if it runs like it should, this will save a lot of time in my case.

The code is available here 

https://github.com/aarizaq/inetmanet-4.x/tree/lora_flora

I have integrate it inside inetmanet-4.x that will facilitate the test

RobsydRosca, 2019-04-10:
Thank you so much for your work porting Inet4 in omnet-5.4.1
I am new in this environment, also, this is not my specific area of interest. At moment, I'm studying the the LoRa reliability mesh networks for applications that require a certain robustness in data transmission.
For this reason I installed omnetpp on Ubuntu 18.04.2 LTS in order to perform some simulations.
It works fine, because I followed the tutorial, compiling and testig the examples without problems.
Than I tried to use FLoRa (Framework for LoRa), but before this I installed inetmanet-4.x.
Unfortunately I am not able to run any examples on inetmanet.
Omnet compiles them with some warnings, but without errors (the warnings refers as "no such paramenters" in the .ned files).
The serious problem arises when the simulation is going to start printing the following error:

> Setting up Qtenv...
> 
> Simulation terminated with exit code: 134
> Working directory: /home/rob/Programs/omnetpp-5.4.1/samples/inetmanet-4.x-master/examples/ethernet/lans
> Command line: opp_run -m -n ../../../src:../..:../../../tutorials:../../../showcases --image-path=../../../images -l ../../../src/INET omnetpp.ini
> 
> Environment variables:
> PATH=/home/rob/Programs/omnetpp-5.4.1/bin::/home/rob/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin:/home/rob/Programs/omnetpp-5.4.1/bin
> LD_LIBRARY_PATH=/home/rob/Programs/omnetpp-5.4.1/lib::/home/rob/Programs/omnetpp-5.4.1/samples/inetmanet-4.x-master/src:
> OMNETPP_IMAGE_PATH=/home/rob/Programs/omnetpp-5.4.1/images

At the same time the following string is printed in the linux shell I used to launch omnetpp:

> *** buffer overflow detected ***: /home/rob/Programs/omnetpp-5.4.1/bin/opp_run terminated 

No other information about the problem are issued.
do you have any suggestions?

p.s.
at the moment it is not useful to tell that, if I install "inetmanet-4.x-lora_flora" I get many errors that do not  allow the compilation of the examples

aarizaq, 2019-04-11:
Have you activated Lora?

You need to select the package,
project  →  properties → OMNeT++ → Project Features and here, you need to select the Lora package.

This version of OMNeT and the code can run in windows as well.

RobsydRosca, 2019-04-11:
Thank you very much Mariusz. I solved that specific problem. And I suddenly
erased my question on github. I ran the simulation in release mode instead
of debug mode.
Unfortunately I still have problems because many errors occurs.
Before to ask help again I would like to be sure I have no other chance.
At moment I am trying to install Castalia, on omnetpp 4.6, but I have
troubles even there.
Sometime I hope somebody release a virtual machine with environments ready
to be used. It is a mess to waste so much time installing SW.
Anyway, thank you again.

aarizaq, 2019-04-12:
Can you send me information about the errors?
If the errors are warnings or errors and in which files are.
inet-framework uses extensively dynamic loading modules in the ned and ini files and the IDE doesn't know the parameters that the module uses and it shows this problem like warnings

RobsydRosca, 2019-04-13:
Thank you very much indeed,
I send you all the information I can collect at moment, hoping to solve my
impasse.
I suppose it is better to avoid a public discussion because my problems
could be trivial, but if you believe, we can move this thread on the web.
Also let me know if you need any kind of further information.

aarizaq, 2019-04-14:
Thanks, without more information I cannot if the error is important or not

RobsydRosca, 2019-04-15:
Hi Mariuszslabicki/Flora
After many operations and attempts I managed to make the simulation works.
Here the window capture of results of the simulation of LoraNetworkTest:
[image: image.png]
Now I got a working system I need to study the structure of LoRa framework
in order to implement my personal network structure.
Please can you suggest me where to get documentation about
inetmanet-4.x-lora_flora?
I would need something like "user guide", reference, manuals, examples and
so on.
I realize that this SW has not been developed by SW houses, but I hope to
find a larger door to enter.
Can you help me?

mariuszslabicki, 2022-06-20:
Hi!

We have published a new version of FLoRa, and added also manual how to setup a basic simulation. You can find more details here:
https://flora.aalto.fi/

closed 2022-06-20 by mariuszslabicki