florasim/flora #11: Documentation

spiroskat, 2019-07-13:
Hi,
I am new user of OMNet++, and i would like to use FLoRa. But studying the source code i find difficult to understand how each function works (so i cannot add for example a new algorithm), as the code is uncommented, and i cannot find the architecture of your project. Could you upload something like UML , and how to use documen ?

mariuszslabicki, 2019-07-13:
Hi
The simple howto can be found on the webpage: https://flora.aalto.fi/

The architecture is based on the OMNeT++ modules architecture, so this is a good starting point if you want to understand the simulation environment.

spiroskat, 2019-08-02:
Hi,
i would like to ask, if i want to implement a logic in a high level, for example in the server i want to send a downlink message in various moments etc and i want to use FLoRa as a black box, i don't care how lora actually works in lower levels, i only want to have a logic running in a lora network, where is recommended to write this code , and is there something like an API (of the FLoRa) that i can use such as for example send_to_lora_nodes(message) ?

spiroskatsampiris, 2020-01-05:
Happy New year, i would like to ask about the energy consumption. I want to do the following: the nodes have some abstract states in high level and each state corresponds to different energy additional energy consumption. The states are changed by messages. The messages start from the nodes, so it follows SimpleLoRaApp to etc, the message arrives to the network server where suppose from state 0 of the node changes the state to 1 by sending downlink message. i need every time that the energy consumpution is calculated, the energy consumption module to understand which state the node is. until now i have errors like variables are not defined in this scope, but despite the fact i have included the headers. More specificallly in SimpleLoRaApp.h in sendJoin Function i add the last line.
`    cInfo->setLoRaTP(loRaTP);
    cInfo->setLoRaCF(loRaCF);
    cInfo->setLoRaSF(loRaSF);
    cInfo->setLoRaBW(loRaBW);
    cInfo->setLoRaCR(loRaCR);
    cInfo->setState(state);`
how can i inform the energy module what state is?
Thank you in advance

mariuszslabicki, 2020-01-24:
Huh, that is a complex description. I think that you may check signals - Energy module receives signals, which determine what is the state.

FireHunter27, 2020-05-26:
I have a question too, in the configuration file, It says **.loRaNodes[*].**initialLoRaCR = 4, What is this ? Is it equal to 4/8 ?

spiroskatsampiris, 2020-07-28:
Hello, i would like to ask if the energy consumption is calculated for the period between the transmissions, or is only calculated for the transmission/reception of the signal. Let say that in every 1000s period a node transmits. The energy consumption in the 1000s period is calculated? Also, i would like to ask if this 1000s period can be used in the energy module of the simulator (if the value set in the configuration file is "visible" in the Energy module). Specifically, i would like to have different energy consumption in some time slots between transmissions. Something like Sleep for 99s->Wake for 1s->Sleep for 99s->Wake And Transmit through LoRa->Sleep->Wake->.....etc

Thank in advance

spiroskatsampiris, 2020-07-28:
@frkngksl  Yes, it means the 4/8

mariuszslabicki, 2022-06-20:
Hi!

We have published a new version of FLoRa, and added also manual how to setup a basic simulation. You can find more details here:
https://flora.aalto.fi/

closed 2022-06-20 by mariuszslabicki