florasim/flora #12: Problem during setup

peggyopal, 2019-09-10:
Hello, I have managed to build INET 3.6.6 and FLoRa 0.8 in OMNeT++ 5.5.1 on an Ubuntu 18.04 machine, however, when I Run As OMNeT++ Simulation I get the following error:

`Cannot evaluate parameter 'initialLoRaTP': operator "*": Refusing to perform computations involving quantities with nonlinear units (3dBm) -- in module (inet::SimpleLoRaApp) LoRaNetworkTest.loRaNodes[0].SimpleLoRaApp (id=30), during network setup`

I have tried using multiple different INET versions, such as 4.1.1, 4.1.0, 3.6.6, 3.6.5, 3.6.4, 3.6.3, but have always had issues with either getting FLoRa to compile with override errors (v4.1.1) or just not being able to compile INET (the rest of the versions).

mariuszslabicki, 2019-09-25:
Hi

Is the "initialLoRaTP" set? It seems that the parameter is not configured in ini file.

peggyopal, 2019-09-25:
Hi, looking at the error above, yes that is what I assume.  What should the parameter be configured as in the ini file?

mariuszslabicki, 2019-09-26:
Please check example ini files. There you can find how nodes can be configured. 

Example ini file is [here](https://github.com/mariuszslabicki/flora/blob/master/simulations/loRaNetworkTest.ini)

leonardonunes3, 2019-10-08:
Hey,

I have faced the same error, i am using OMNet++ 5.4.1 with INET 3.6.6 and FloRa 0.8 and i get: 

<!> Cannot evaluate parameter 'initialLoRaTP': operator "*": Refusing to perform computations involving quantities with nonlinear units (3dBm) -- in module (inet::SimpleLoRaApp) LoRaNetworkTest.loRaNodes[0].SimpleLoRaApp (id=30), during network setup

when try to run  loRaNetworkTest.ini. I have already checked the ini file, and it's equal. Any suggestions to solve this?

mariuszslabicki, 2019-10-08:
It seems the simulation is not run correctly as the OMNeT cannot find the "initialLoRaTP" parameter. I am not sure if the FLoRa can works with this omnet/inet version. But it looks like the simulation is run inproperly. I suggest to double-check everything.

mariuszslabicki, 2020-01-24:
Let me close it as it is rather old topic. Please reopen if needed.

closed 2020-01-24 by mariuszslabicki