florasim/flora #13: Syntax errors

Smangaliso1, 2019-09-19:
Hi, I wanna build Flora, however after doing "build project", error with below log occurred. I am using omnetpp 5.5.1 and inet-4.1.1  in ubuntu 18.10. I had plenty of errors but manage to solve them though I'm still new in omnetpp and c++.
08:06:06 **** Incremental Build of configuration release for project Flora ****
make MODE=release all 
src/LoRa/LoRaGWMac.cc
In file included from src/LoRa/LoRaGWMac.cc:16:
src/LoRa/LoRaGWMac.h:38:29: error: ‘virtual inet::InterfaceEntry* inet::LoRaGWMac::createInterfaceEntry()’ marked ‘override’, but does not override
     virtual InterfaceEntry *createInterfaceEntry() override;
                             ^~~~~~~~~~~~~~~~~~~~
src/LoRa/LoRaGWMac.h:42:18: error: ‘virtual void inet::LoRaGWMac::handleUpperPacket(omnetpp::cPacket*)’ marked ‘override’, but does not override
     virtual void handleUpperPacket(cPacket *msg) override;
                  ^~~~~~~~~~~~~~~~~
src/LoRa/LoRaGWMac.h:43:18: error: ‘virtual void inet::LoRaGWMac::handleLowerPacket(omnetpp::cPacket*)’ marked ‘override’, but does not override
     virtual void handleLowerPacket(cPacket *msg) override;
                  ^~~~~~~~~~~~~~~~~
In file included from /home/osboxes/Downloads/omnetpp-5.5.1/include/omnetpp/cobject.h:26,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/include/omnetpp/cnamedobject.h:23,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/include/omnetpp/cownedobject.h:23,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/include/omnetpp/carray.h:19,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/include/omnetpp.h:28,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/precompiled.h:28,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/precompiled_release.h:1,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/INETDefs.h:23,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/physicallayer/contract/packetlevel/IAntenna.h:21,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/physicallayer/contract/packetlevel/IRadio.h:21,
                 from src/LoRa/LoRaGWMac.h:20,
                 from src/LoRa/LoRaGWMac.cc:16:
src/LoRa/LoRaGWMac.cc: In function ‘omnetpp::cObject* inet::__factoryfunc_23()’:
src/LoRa/LoRaGWMac.cc:23:15: error: invalid new-expression of abstract class type ‘inet::LoRaGWMac’
 Define_Module(LoRaGWMac);
               ^~~~~~~~~
In file included from src/LoRa/LoRaGWMac.cc:16:
src/LoRa/LoRaGWMac.h:32:7: note:   because the following virtual functions are pure within ‘inet::LoRaGWMac’:
 class LoRaGWMac: public MacProtocolBase {
       ^~~~~~~~~
In file included from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/LayeredProtocolBase.h:21,
                 from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/linklayer/base/MacProtocolBase.h:21,
                 from src/LoRa/LoRaGWMac.h:22,
                 from src/LoRa/LoRaGWMac.cc:16:
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/lifecycle/OperationalBase.h:60:18: note: 	‘virtual void inet::OperationalBase::handleStartOperation(inet::LifecycleOperation*)’
     virtual void handleStartOperation(LifecycleOperation *operation) = 0;
                  ^~~~~~~~~~~~~~~~~~~~
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/lifecycle/OperationalBase.h:61:18: note: 	‘virtual void inet::OperationalBase::handleStopOperation(inet::LifecycleOperation*)’
     virtual void handleStopOperation(LifecycleOperation *operation) = 0;
                  ^~~~~~~~~~~~~~~~~~~
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/common/lifecycle/OperationalBase.h:62:18: note: 	‘virtual void inet::OperationalBase::handleCrashOperation(inet::LifecycleOperation*)’
     virtual void handleCrashOperation(LifecycleOperation *operation) = 0;
                  ^~~~~~~~~~~~~~~~~~~~
In file included from src/LoRa/LoRaGWMac.h:22,
                 from src/LoRa/LoRaGWMac.cc:16:
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/linklayer/base/MacProtocolBase.h:46:18: note: 	‘virtual void inet::MacProtocolBase::configureInterfaceEntry()’
     virtual void configureInterfaceEntry() = 0;
                  ^~~~~~~~~~~~~~~~~~~~~~~
src/LoRa/LoRaGWMac.cc: In member function ‘virtual inet::InterfaceEntry* inet::LoRaGWMac::createInterfaceEntry()’:
src/LoRa/LoRaGWMac.cc:63:48: error: no matching function for call to ‘inet::InterfaceEntry::InterfaceEntry(inet::LoRaGWMac*)’
     InterfaceEntry *e = new InterfaceEntry(this);
                                                ^
In file included from /home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/linklayer/base/MacProtocolBase.h:23,
                 from src/LoRa/LoRaGWMac.h:22,
                 from src/LoRa/LoRaGWMac.cc:16:
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/networklayer/common/InterfaceEntry.h:154:5: note: candidate: ‘inet::InterfaceEntry::InterfaceEntry()’
     InterfaceEntry();
     ^~~~~~~~~~~~~~
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/networklayer/common/InterfaceEntry.h:154:5: note:   candidate expects 0 arguments, 1 provided
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/networklayer/common/InterfaceEntry.h:119:5: note: candidate: ‘inet::InterfaceEntry::InterfaceEntry(const inet::InterfaceEntry&)’
     InterfaceEntry(const InterfaceEntry& obj);
     ^~~~~~~~~~~~~~
/home/osboxes/Downloads/omnetpp-5.5.1/samples/inet/src/inet/networklayer/common/InterfaceEntry.h:119:5: note:   no known conversion for argument 1 from ‘inet::LoRaGWMac*’ to ‘const inet::InterfaceEntry&’
make: *** [Makefile:140: out/gcc-release//src/LoRa/LoRaGWMac.o] Error 1
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

08:06:08 Build Failed. 6 errors, 0 warnings. (took 2s.355ms)
what is the problem?
Your assistance will be highly appreciated.

Smangaliso1, 2019-09-19:
@mariuszslabicki

mariuszslabicki, 2019-09-25:
Hi

Please try with versions mentioned in the website: https://flora.aalto.fi/ 

As far as I remember, there was an update in new inet, and flora is not updated with it.

Smangaliso1, 2019-09-25:
Thank you, I'll try and let you know afterwards.

Smangaliso1, 2019-09-27:
Hi 
Thank you initial problem fixed!!!

However, I've also encountered another bug while testing Flora project.

<!> Cannot evaluate parameter 'initialLoRaTP': operator "*": Refusing to perform computations involving quantities with nonlinear units (3dBm) -- in module (inet::SimpleLoRaApp) LoRaNetworkTest.loRaNodes[0].SimpleLoRaApp (id=30), during network setup

mariuszslabicki, 2019-09-27:
Could you paste the config file?

Smangaliso1, 2019-09-27:
Sorry, config file for Omnetpp or Flora I'm not sure in that case.

mariuszslabicki, 2019-09-27:
The .ini file which defines parameters for nodes. If you have run the example simulation from repository, the configuration is set in the "loRaNetworkTest.ini" file.

Smangaliso1, 2019-09-27:
Ok, I get you now. below I just copied what is in the file.
[General]
network = LoRaNetworkTest
rng-class = "cMersenneTwister"

# network features
**.numberOfGateways = 1
**.loRaGW[0].numUdpApps = 1
**.loRaGW[0].packetForwarder.localPort = 2000
**.loRaGW[0].packetForwarder.destPort = 1000
**.loRaGW[0].packetForwarder.destAddresses = "networkServer"
**.loRaGW[0].packetForwarder.indexNumber = 0

**.loRaGW[1].numUdpApps = 1
**.loRaGW[1].packetForwarder.localPort = 2000
**.loRaGW[1].packetForwarder.destPort = 1000
**.loRaGW[1].packetForwarder.destAddresses = "networkServer"
**.loRaGW[1].packetForwarder.indexNumber = 1

**.networkServer.numUdpApps = 1
**.networkServer.**.evaluateADRinServer = false
**.networkServer.udpApp[0].typename = "NetworkServerApp"
**.networkServer.udpApp[0].destAddresses = "loRaGW[0]"
**.networkServer.udpApp[0].destPort = 2000
**.networkServer.udpApp[0].localPort = 1000
**.networkServer.udpApp[0].adrMethod = ${"avg"}

**.numberOfNodes = 10
**.numberOfPacketsToSend = 0 #0 means infinite number of packets
sim-time-limit = 7d
warmup-period = 1d
simtime-resolution = -11

**.timeToFirstPacket = exponential(100s)
**.timeToNextPacket = exponential(100s)
**.alohaChannelModel = false

#nodes features
**.loRaNodes[*].**.initFromDisplayString = false
**.loRaNodes[*].**.evaluateADRinNode = false
**.loRaNodes[*].**initialLoRaSF = intuniform(7,12)
**.loRaNodes[*].**initialLoRaBW = 125 kHz
**.loRaNodes[*].**initialLoRaCR = 4
**.loRaNodes[*].**initialLoRaTP = (2dBm + 3dBm*intuniform(0, 4))

# deployment of nodes in a circle with radius=maxGatewayDistance and gateway at gatewayX,gatewayY
#**.loRaNodes[*].deploymentType = "circle"
#**.loRaNodes[*].maxGatewayDistance = 120.0
#**.loRaNodes[*].gatewayX = 240
#**.loRaNodes[*].gatewayY = 240

# random deployment of nodes in a square square area
**.loRaNodes[*].**.initialX = uniform(0m, 480m)
**.loRaNodes[*].**.initialY = uniform(0m, 480m)

#gateway features
**.LoRaGWNic.radio.iAmGateway = true
**.loRaGW[*].**.initFromDisplayString = false
**.loRaGW[0].**.initialX = 240m#uniform(0m, 50m)
**.loRaGW[0].**.initialY = 240m

#power consumption features
**.loRaNodes[*].LoRaNic.radio.energyConsumerType = "LoRaEnergyConsumer"
**.loRaNodes[*].**.energySourceModule = "IdealEpEnergyStorage"
**.loRaNodes[*].LoRaNic.radio.energyConsumer.configFile = xmldoc("energyConsumptionParameters.xml")

#general features
**.sigma = 0
**.constraintAreaMinX = 0m
**.constraintAreaMinY = 0m
**.constraintAreaMinZ = 0m
**.constraintAreaMaxX = 480m
**.constraintAreaMaxY = 480m
**.constraintAreaMaxZ = 0m

LoRaNetworkTest.**.radio.separateTransmissionParts = false
LoRaNetworkTest.**.radio.separateReceptionParts = false

**.delayer.config = xmldoc("cloudDelays.xml")
**.radio.radioMediumModule = "LoRaMedium"
**.LoRaMedium.pathLossType = "LoRaLogNormalShadowing"
**.minInterferenceTime = 0s
**.displayAddresses = false

mariuszslabicki, 2019-09-27:
Are you sure that this is the config file that is evaluated in your simulation? From the error message: annot evaluate parameter 'initialLoRaTP': operator "*" it seems that the "initialLoRaTP" is not set. But the config file has all parameters configured.

BTW: have you evaluated example simulation?

Smangaliso1, 2019-10-01:
Hey,
I managed to fix the error, All was need is to remove this operator "*" from the config files .in.

Thank you.

closed 2019-10-01 by mariuszslabicki