florasim/flora #14: Problem with building of Flora

kamilksiazek, 2019-09-26:
Hello.

I have a problem connected with building Flora (I have OMNeT++ 5.4.1). I tried build project with INET 3.6.6 and I got some errors:

`src/LoRa/LoRaRadio.cc:33:24: error: definition of dllimport static field not allowed`
`src/LoRa/LoRaRadio.cc:34:24: error: definition of dllimport static field not allowed`
`src/LoRa/LoRaRadio.cc:35:24: error: definition of dllimport static field not allowed`
`src/LoRa/LoRaRadio.cc:36:24: error: definition of dllimport static field not allowed`
`src/LoRa/LoRaRadio.cc:37:24: error: definition of dllimport static field not allowed`
`make: *** [Makefile:155: out/clang-release//src/LoRa/LoRaRadio.o] Error 1`

In a configure.user I set 
`PREFER_CLANG=no`
according to your instruction.

Similar errors occurred with INET 4.1.1 and INET 3.5.0. I have Windows 10. Could you help me to solve this problem?

Thank you very much.

mariuszslabicki, 2019-09-27:
I am sorry, it is not tested in Windows. I know from other people that it doesn't work in this environment, but I don't have resources to debug it.

kamilksiazek, 2019-10-07:
Ok, thank you for your reply. I installed Flora on Ubuntu already.

closed 2020-01-24 by mariuszslabicki