florasim/flora #17: Comparing Flora and NS 3 for project

adnan21471, 2019-12-24:
Hello..I am new to inet and Ns3.
I am currently deciding which to choose over other meaning how is Flora better than Ns 3 (by features) and vice versa..Also which is easy to learn quickly.

Would really appreciate if someone could guide me..I am not focusing on machine learning, but just resource allocation problems..Have a nice day

spiroskatsampiris, 2020-01-09:
Each environment have advantages and disadvantages. Though FLoRa is quiet good and powerful giving end to end simulations with great visualization. You can google something like comparison of lora simulation environments for more details.

mariuszslabicki, 2020-01-24:
To be honest, I have no idea. We are using FLoRa in our research as it fits our needs, and I am not aware what are capabilities of other simulators.

closed 2020-01-29 by mariuszslabicki