florasim/flora #18: Plotting statistics with FLoRa

spiroskatsampiris, 2020-01-14:
Hi,
i would like to ask, what is the best way to plot statistics in the way you presented the results in the figures of the paper "Adaptive Configuration of LoRa Networks for Dense IoT Deployments". Specifically, plots like the figures 3,5,5,6. Did you use something built-in in the Omnet or the analysis was done using python?

Thank you in advance,

mariuszslabicki, 2020-01-24:
As far as I remember, it was done in Python (with matplotlib). We have parsed data from omnet files, and then loaded them into python scripts to plot graphs.

spiroskatsampiris, 2020-01-25:
Thanks for the answer, i would like to ask something that caught my attention. In LoRaPhy directory, there are some path loss models. But apart from the path loss model from the paper http://doi.acm.org/10.1145/2988287.2989163, where are specified what are the parameters, there is a model called Oulu path loss. Could you provide the publication where the model was calculated? Also, it looks to be the same with the paper's "Do lora low-power wide-area networks scale" with this part " - antennaGain ". Could you elaborate a little bit about these path losses that are not included in your publication ?

Thank you in advance

mariuszslabicki, 2020-01-29:
As I remember, the second model was based on: 
http://cc.oulu.fi/~kmikhayl/site-assets/pdfs/2015_ITST.pdf

But we have not used it in the paper.

closed 2020-01-30 by mariuszslabicki