florasim/flora #19: New Routing Protocols

FireHunter27, 2020-02-28:
How can we modify this framework to test new routing protocols for LoRa devices ?

mariuszslabicki, 2020-02-28:
Well, you can implement the routing protocol.

mariuszslabicki, 2020-02-28:
But just to be sure - what do you mean by that? Normal LoRa use case is "send data from end node to GW, and then it is transmitted by more reliable link to Network Server". At which stage you see space to add routing protocol?

FireHunter27, 2020-02-28:
Actually, I have a research project about smart agriculture, and I want to simulate a WSN environment that using LoRa modules. I know that ,as you said, normal LoRaWAN case uses bidirectional communication between an end node and gateway. However, one of my research question is can we decrease power consumption and minimize the setup cost using node to node communication. For example, since gateway' hardware cost is more expensive than an end node cost, can we use the node to node communication for transmission of data for nodes that are away from gateway (like mesh networks). Clearly, your simulation provides power consumption simulations (by the way thank you for such a great framework), and I want to learn how I can implement this in the framework.

mariuszslabicki, 2020-03-02:
Hi

Well, the research question is interesting. Please keep in mind that there are many "if" in it, so you need to clearly set up assumptions. The first issue that I have at the top of my head are:

* LoRa uses that small amount of energy because most of the time devices are in sleep mode (at least radio part in class A). If you allow device2device communication, how they will know when to listen to another device?  Listen constantly, periodical or anyhow else?

You should be able to simulate that in FLoRa, but it need to check at least MAC layer.

FireHunter27, 2020-03-02:
Hi

Thank you for your interest and question. We will make assumption about some "if's", and you can be sure that at the end of paper, this amazing question will be answered. We actually simualte the network for this type of questions, and different scenarios (like dead node connection). But I don't understand the  meaning of "it need to check at least MAC layer." statement. In the current configurations of FLoRa, is it enough to change only LoRaMac module for simulating device2device communication?

mariuszslabicki, 2020-03-03:
Well, current FLoRa works in compliance (more or less) with LoRaWAN. Which assumes that you transmit data from endnodes by GW to network server. You want to change this behavior significantly. 

LoRaMAC controls finite state machine which controls radio state. Then probably you need to update also network layer and application layer.

closed 2020-03-03 by mariuszslabicki