florasim/flora #21: Computing error with FLORA

DOMCER-37, 2020-03-24:
Hi,

I manage to compile FLORA by changing the version of INET I used inet-3.6.7, but now that the compilation works i have trouble during the computation , here is what I obtanined in the error window Cannot evaluate parameter 'initialLoRaTP': operator "*": Refusing to perform computations involving quantities with nonlinear units (3dBm) -- in module (inet::SimpleLoRaApp) LoRaNetworkTest.loRaNodes[0].SimpleLoRaApp (id=30), during network setup

I read that you already discuss with another people but I really don't understand how the error can be corrected..I am very sorry again and many thanks for your help 

DOCCER

mariuszslabicki, 2020-03-25:
Hi

Could you verify if you have set everything in ini files? 

From the definition in ned file for simple lora app:
double initialLoRaTP @unit(dBm) = default(14dBm);

It means that initial lora tp value must be set in dBm.

DOMCER-37, 2020-03-25:
Hi,

first for your time and consideration, really and nice to meet you, by email…
Just for your information, I teach in a school Electrical Engineering and would like to introduce IoT networks through modeling, this is why I try to take in hand OMNET, INET and of course your software, Flora.

First, I manage to compile OMNET, INET and finally Flora, even id many many many warnings appeared.

I launched the first example LoraNetworkTest.inI (see attached file) and the following error occurred :"Cannot evaluate parameter 'initialLoRaTP': operator "*": Refusing to perform computations involving quantities with nonlinear units (3dBm) -- in module (inet::SimpleLoRaApp) LoRaNetworkTest.loRaNodes[0].SimpleLoRaApp (id=30), during network setup ».

I have looked at the SimpleLoraApp.ned and it seems that the definition in ned file was well since the default unit was dBm (here is the file).

To manage to launch the software, I suppressed * in the .ini file as you can see in the .docx document lines 41 42 43

And it works ! why I don’t know…

I now test the following example LoraNetwork_skeleton.ini (see attached file) and now a new error appeared :"Mobility position is not a finite number after initialize (x=inf,y=inf,z=0) -- in module (inet::StationaryMobility) LoRaNetworkTest.loRaNodes[0].mobility (id=19), during network initialization »

Perhaps you will helped me to solve the problems. Just I am a neophyte in this kind of working environment (unfortunately it’s not the heart of my scientific activity), this is why I am lost.

Again many many thanks for yours councils.

Best

Dominique CERTON

mariuszslabicki, 2020-03-26:
The skeleton file is just a skeleton ini with partial network configuration. Still you need to extend this simulation with another .ini file which add rest of needed configuration. We used that in batch simulations, when we changed only topology parameters, but network configuration was the same.

closed 2020-03-27 by mariuszslabicki

DOMCER-37, 2020-03-27:
That’s works many thanks !

It will be my last questions, have you a summarize of the parameters and functions that can be changed ?

I guess that by default all is in the .ini files ?

Thanks
Best

> 
> 
> The skeleton file is just a skeleton ini with partial network configuration. Still you need to extend this simulation with another .ini file which add rest of needed configuration. We used that in batch simulations, when we changed only topology parameters, but network configuration was the same.

Dominique CERTON
Resp. Dpt Electronique et Energie
Polytech TOURS
dominique.certon@univ-tours.fr

mariuszslabicki, 2020-04-03:
Hi. I don't have this kind of summary. Probably you need to check it in code directly.