florasim/flora #22: updateTransceiverState and updateTransceiverPart

FireHunter27, 2020-03-26:
Hello sir,

I was observing the implementation and I encountered these two function in LoRaRadio.cc. As far as I understood, these ones are for MAC layer in order to change RadioState. Is there any usage of these functions beside of that? Also, I can't figure out the difference between RadioReceptionStates. What do RECEPTION_STATE_BUSY and RECEPTION_STATE_UNDEFINED states represent? Thank you for consideration.

mariuszslabicki, 2020-03-26:
Hi. Yes, you are right, they are to deal with Radio State. The state RECEPTION_STATE_UNDEFINED is set when we cannot say what is the reception state - for example, when radio is transmitting. I don't remember the whole logic from the top of my head, you need to go through this file.

FireHunter27, 2020-03-26:
Thank you for reply, I'll close the issue

closed 2020-03-26 by FireHunter27