florasim/flora #28: Build Project Eror

danzuha, 2020-11-09:
Hi Marius,

I get an error when i want to buld project, it happen at the last step on the build this cause me unable to run simulation. 
I am using windows 10. Hope you can help me to resolve this. The error is as follows. for now i preffer gcc over clang. 

Creating executable: ../out/clang-release/src/flora.exe
../out/clang-release/src/LoRa/LoRaGWMac.o:(.rdata+0x8d0): undefined reference to `typeinfo for inet::MACProtocolBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZN4inet13physicallayer13FlatRadioBaseD1Ev]+0x3): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZN4inet13physicallayer13FlatRadioBaseD0Ev]+0xb): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0xce0): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x14e0): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x1570): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x15d0): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x1dc0): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x1e50): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x1eb0): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x2688): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x2718): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.rdata+0x27a0): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTv0_n24_N4inet13physicallayer13FlatRadioBaseD1Ev]+0xa): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTv0_n24_N4inet13physicallayer13FlatRadioBaseD0Ev]+0x12): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTvn8_n24_N4inet13physicallayer13FlatRadioBaseD1Ev]+0xf): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTvn8_n24_N4inet13physicallayer13FlatRadioBaseD0Ev]+0x17): undefined reference to `VTT for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZN4inet13physicallayer19NarrowbandRadioBaseD1Ev]+0x3): undefined reference to `VTT for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZN4inet13physicallayer19NarrowbandRadioBaseD0Ev]+0xb): undefined reference to `VTT for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTv0_n24_N4inet13physicallayer19NarrowbandRadioBaseD1Ev]+0xa): undefined reference to `VTT for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTv0_n24_N4inet13physicallayer19NarrowbandRadioBaseD0Ev]+0x12): undefined reference to `VTT for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTvn8_n24_N4inet13physicallayer19NarrowbandRadioBaseD1Ev]+0xf): undefined reference to `VTT for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaGWRadio.o:(.text[_ZTvn8_n24_N4inet13physicallayer19NarrowbandRadioBaseD0Ev]+0x17): more undefined references to `VTT for inet::physicallayer::NarrowbandRadioBase' follow
../out/clang-release/src/LoRa/LoRaMac.o:(.rdata+0xba0): undefined reference to `typeinfo for inet::MACProtocolBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x1230): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x1a30): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x1ac0): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x1b20): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x2310): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x23a0): undefined reference to `typeinfo for inet::physicallayer::NarrowbandRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x2400): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x2bd8): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x2c68): undefined reference to `typeinfo for inet::physicallayer::Radio'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata+0x2cf0): undefined reference to `typeinfo for inet::physicallayer::FlatRadioBase'
../out/clang-release/src/LoRa/LoRaRadio.o:(.text[_ZN7omnetpp14check_and_castIPN4inet13physicallayer10RadioFrameENS_8cMessageEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::physicallayer::RadioFrame'
../out/clang-release/src/LoRa/LoRaRadio.o:(.text[_ZN7omnetpp14check_and_castIPN4inet13physicallayer21ConfigureRadioCommandENS_7cObjectEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::physicallayer::ConfigureRadioCommand'
../out/clang-release/src/LoRa/LoRaRadio.o:(.text[_ZN7omnetpp14check_and_castIPN4inet13physicallayer10RadioFrameENS2_11IRadioFrameEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::physicallayer::RadioFrame'
../out/clang-release/src/LoRa/LoRaRadio.o:(.text[_ZN7omnetpp14check_and_castIPKN4inet13physicallayer19ReceptionIndicationENS_7cObjectEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::physicallayer::ReceptionIndication'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata[_ZTIPN4inet13physicallayer10RadioFrameE]+0x18): undefined reference to `typeinfo for inet::physicallayer::RadioFrame'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata[_ZTIPN4inet13physicallayer21ConfigureRadioCommandE]+0x18): undefined reference to `typeinfo for inet::physicallayer::ConfigureRadioCommand'
../out/clang-release/src/LoRa/LoRaRadio.o:(.rdata[_ZTIPKN4inet13physicallayer19ReceptionIndicationE]+0x18): undefined reference to `typeinfo for inet::physicallayer::ReceptionIndication'
../out/clang-release/src/LoRa/NetworkServerApp.o:(.text[_ZN7omnetpp14check_and_castIPN4inet17UDPDataIndicationENS_7cObjectEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::UDPDataIndication'
../out/clang-release/src/LoRa/NetworkServerApp.o:(.rdata[_ZTIPN4inet17UDPDataIndicationE]+0x18): undefined reference to `typeinfo for inet::UDPDataIndication'
../out/clang-release/src/LoRa/PacketForwarder.o:(.text[_ZN7omnetpp14check_and_castIPN4inet13physicallayer19ReceptionIndicationENS_7cObjectEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::physicallayer::ReceptionIndication'
../out/clang-release/src/LoRa/PacketForwarder.o:(.rdata[_ZTIPN4inet13physicallayer19ReceptionIndicationE]+0x18): undefined reference to `typeinfo for inet::physicallayer::ReceptionIndication'
../out/clang-release/src/LoRaApp/SimpleLoRaApp.o:(.text+0x2d5): undefined reference to `typeinfo for inet::NodeStatus'
../out/clang-release/src/LoRaApp/SimpleLoRaApp.o:(.text[_ZN7omnetpp14check_and_castIPN4inet18StationaryMobilityENS_7cModuleEEET_PT0_]+0x18): undefined reference to `typeinfo for inet::StationaryMobility'
../out/clang-release/src/LoRaApp/SimpleLoRaApp.o:(.rdata[_ZTIPN4inet18StationaryMobilityE]+0x18): undefined reference to `typeinfo for inet::StationaryMobility'
../out/clang-release/src/LoRaEnergyModules/LoRaEnergyConsumer.o:(.text+0x28d): undefined reference to `typeinfo for inet::power::IdealEpEnergyStorage'
../out/clang-release/src/LoRaEnergyModules/LoRaEnergyConsumer.o:(.rdata+0xb40): undefined reference to `typeinfo for inet::physicallayer::StateBasedEpEnergyConsumer'
[... cut ...]
../out/clang-release/src/LoRaPhy/LoRaTransmitter.o:(.rdata+0x13a0): undefined reference to `typeinfo for inet::physicallayer::NarrowbandTransmitterBase'
../out/clang-release/src/LoRaPhy/LoRaTransmitter.o:(.rdata+0x19f0): undefined reference to `typeinfo for inet::physicallayer::FlatTransmitterBase'
clang++.exe: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:143: ../out/clang-release/src/flora.exe] Error 1
make[1]: Leaving directory '/c/Users/User/Documents/omnetpp-5.2.1/samples/FLORA/flora-0.8/src'
make: *** [Makefile:2: all] Error 2

19:42:36 Build Finished (took 3s.854ms)

danzuha, 2020-11-09:
Hi @mariuszslabicki may I know if it is actually possible to install flora on windows 10?

Vaseem-54, 2020-11-14:
it will not work on windows 10, use Ubuntu 16.4LT, INET 3.6.3 frame work.

rogerpueyo, 2020-11-14:
It also works nicely with Debian Jessie or Debian Buster, using OMNet++ 5.2.1 from [here](https://omnetpp.org/download/old.html) and INET 3.6.4 from [here](https://github.com/inet-framework/inet/releases/tag/v3.6.4).

danzuha, 2020-11-14:
Thanks guys, ill try with ubuntu16.04lt

closed 2020-12-09 by danzuha