florasim/flora #31: How to make a graph for all end nodes energy

danzuha, 2020-12-09:
Hi guys, 

Im new to this omnett, anyone can help me how to get energy consumed by the end nodes? Or the energy for end nodes? Currently im using the sample  file .ini i need to collect them and make a graph for the simulation.

danzuha, 2020-12-21:
Perhaps im not checking at the right place?

gpremsan, 2020-12-21:
The energy consumed is stored in the variable `totalEnergyConsumed` for each node. For example, if there's one node in the network, the total energy spent in sending / receiving messages is stored in `scalar LoRaNetworkTest.loRaNodes[0].LoRaNic.radio.energyConsumer totalEnergyConsumed` in the scalar output file. The unit is Joules.

danzuha, 2020-12-22:
> The energy consumed is stored in the variable `totalEnergyConsumed` for each node. For example, if there's one node in the network, the total energy spent in sending / receiving messages is stored [...]

Hi gpremsan 

Thanks for the replies, really appreciate it. May I know if where can i get the totalEnergyconsumed scalar? Is there specific file or in GUI? I think i cannot find any match in GUI

fyi, im using the loRaNetworkTest.ini to get to use to this.

mariuszslabicki, 2020-12-22:
You need to check scalar output files. They are saved somewhere in the simulation directory, not remember exact path from the top of my head.

danzuha, 2020-12-23:
Hi @mariuszslabicki 

Thanks for the info, just for future refference incase of anyone who had same issue as me, here is the file path to get the info![Uploading 103044288...]

danzuha, 2020-12-23:
Got it, thanks!

closed 2020-12-24 by danzuha