florasim/flora #35: FLoRa build fails with several errors

smblackledge, 2021-02-14:
Nine errors and six warnings from the build that automatically happened as part of the "import" menu item.

Windows 10 Pro, Omnet++ 5.6.2, Inet 4.2, and FLoRa 0.8. The build output is included below.

There are some files in the inet\misc\rename folder (see below) that list either renames that have been done or should be done; I'm not sure which.  I also don't know if there's already a tool distributed with FLoRa or inet (inet\migrate folder?) that would do the renames? I do know, however, that manually renaming the invalid names allow other errors to appear.

FLoRa is not in the workspace folder (samples), Maybe that's caused by the failed build.  I'm not sure what happens to the source files in the FLoRa\src distribution, although there is a new folder named "out" which may contain some of the build errors.

A regression build by the developers/maintainers is needed.

Rename-related files in inet\misc\rename:

- class_renames_cumulated.txt
- replacements.pl
- class_renames_cumulated.txt
- migrateinclude.pl
- suggested_class_renames.txt

19:47:11 **** Build of configuration gcc-debug for project flora ****
make MODE=debug
make[1]: Entering directory '/c/Users/Scott/Documents/flora/src'
MSGC: LoRa/LoRaMacControlInfo.msg
MSGC: LoRa/LoRaMacFrame.msg
MSGC: LoRaApp/LoRaAppPacket.msg
MSGC: LoRaPhy/LoRaRadioControlInfo.msg
LoRa/LoRaGWMac.cc
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:20:10: warning: non-portable path to file '"inet/linklayer/contract/IMacProtocol.h"'; specified path differs in case from file name on disk [-Wnonportable-include-path]
#include "inet/linklayer/contract/IMACProtocol.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         "inet/linklayer/contract/IMacProtocol.h"
LoRa/LoRaGWMac.h:21:10: warning: non-portable path to file '"inet/linklayer/base/MacProtocolBase.h"'; specified path differs in case from file name on disk [-Wnonportable-include-path]
#include "inet/linklayer/base/MACProtocolBase.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         "inet/linklayer/base/MacProtocolBase.h"
LoRa/LoRaGWMac.h:31:25: error: unknown class name 'MACProtocolBase'; did you mean 'MacProtocolBase'?
class LoRaGWMac: public MACProtocolBase {
                        ^~~~~~~~~~~~~~~
                        MacProtocolBase
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/linklayer/base/MACProtocolBase.h:29:16: note: 'MacProtocolBase' declared here
class INET_API MacProtocolBase : public LayeredProtocolBase, public cListener
               ^
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:37:29: error: 'createInterfaceEntry' marked 'override' but does not override any member functions
    virtual InterfaceEntry *createInterfaceEntry() override;
                            ^
LoRa/LoRaGWMac.h:41:18: error: 'handleUpperPacket' marked 'override' but does not override any member functions
    virtual void handleUpperPacket(cPacket *msg) override;
                 ^
LoRa/LoRaGWMac.h:42:18: error: 'handleLowerPacket' marked 'override' but does not override any member functions
    virtual void handleLowerPacket(cPacket *msg) override;
                 ^
LoRa/LoRaGWMac.h:36:18: warning: 'inet::LoRaGWMac::finish' hides overloaded virtual function [-Woverloaded-virtual]
    virtual void finish() override;
                 ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/include\omnetpp/clistener.h:118:18: note: hidden overloaded virtual function 'omnetpp::cIListener::finish' declared here: different number of parameters (2 vs 0)
    virtual void finish(cComponent *component, simsignal_t signalID) {}
                 ^
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:41:18: warning: 'inet::LoRaGWMac::handleUpperPacket' hides overloaded virtual function [-Woverloaded-virtual]
    virtual void handleUpperPacket(cPacket *msg) override;
                 ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/common/LayeredProtocolBase.h:38:18: note: hidden overloaded virtual function 'inet::LayeredProtocolBase::handleUpperPacket' declared here: type mismatch at 1st parameter ('inet::Packet *' vs 'omnetpp::cPacket *')
    virtual void handleUpperPacket(Packet *packet);
                 ^
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:42:18: warning: 'inet::LoRaGWMac::handleLowerPacket' hides overloaded virtual function [-Woverloaded-virtual]
    virtual void handleLowerPacket(cPacket *msg) override;
                 ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/common/LayeredProtocolBase.h:39:18: note: hidden overloaded virtual function 'inet::LayeredProtocolBase::handleLowerPacket' declared here: type mismatch at 1st parameter ('inet::Packet *' vs 'omnetpp::cPacket *')
    virtual void handleLowerPacket(Packet *packet);
                 ^
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:55:18: warning: 'inet::LoRaGWMac::receiveSignal' hides overloaded virtual function [-Woverloaded-virtual]
    virtual void receiveSignal(cComponent *source, simsignal_t signalID, long value, cObject *details) override;
                 ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/linklayer/base/MACProtocolBase.h:91:18: note: hidden overloaded virtual function 'inet::MacProtocolBase::receiveSignal' declared here: type mismatch at 3rd parameter ('omnetpp::cObject *' vs 'long')
    virtual void receiveSignal(cComponent *source, simsignal_t signalID, cObject *obj, cObject *details) override;
                 ^
LoRa/LoRaGWMac.cc:23:15: error: allocating an object of abstract class type 'inet::LoRaGWMac'
Define_Module(LoRaGWMac);
              ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/linklayer/base/MACProtocolBase.h:61:18: note: unimplemented pure virtual method 'configureInterfaceEntry' in 'LoRaGWMac'
    virtual void configureInterfaceEntry() = 0;
                 ^
LoRa/LoRaGWMac.cc:27:5: error: use of undeclared identifier 'MACProtocolBase'; did you mean 'MacProtocolBase'?
    MACProtocolBase::initialize(stage);
    ^~~~~~~~~~~~~~~
    MacProtocolBase
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/linklayer/base/MACProtocolBase.h:29:16: note: 'MacProtocolBase' declared here
class INET_API MacProtocolBase : public LayeredProtocolBase, public cListener
               ^
LoRa/LoRaGWMac.cc:63:29: error: no matching constructor for initialization of 'inet::InterfaceEntry'
    InterfaceEntry *e = new InterfaceEntry(this);
                            ^              ~~~~
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/networklayer/common/InterfaceEntry.h:120:5: note: candidate constructor not viable: no known conversion from 'inet::LoRaGWMac *' to 'const inet::InterfaceEntry' for 1st argument
    InterfaceEntry(const InterfaceEntry& obj);
    ^
C:/Users/Scott/Desktop/omnetpp-5.6.2/samples/inet/src\inet/networklayer/common/InterfaceEntry.h:155:5: note: candidate constructor not viable: requires 0 arguments, but 1 was provided
    InterfaceEntry();
    ^
6 warnings and 7 errors generated.
make[1]: *** [Makefile:152: ../out/clang-debug/src/LoRa/LoRaGWMac.o] Error 1
make[1]: Leaving directory '/c/Users/Scott/Documents/flora/src'
make: *** [Makefile:2: all] Error 2
"make MODE=debug all" terminated with exit code 2. Build might be incomplete.

19:47:14 Build Failed. 9 errors, 6 warnings. (took 3s.172ms)

smblackledge, 2021-02-19:
I closed the FLoRa project in the omnetpp ide, then right-clicked on the "flora" project and deleted it.  I downloaded the "inet3to4" .zip file to get the latest code.  **Please let me know if this isn't the latest code.**  Next, I imported inet3to4 as directed in the flora install document.  Lastly, right-clicked on flora and selected "Build Project."  It looks like most, if not all of the variable renames were fixed, but the build still failed.

17:43:13 **** Build of configuration gcc-debug for project flora ****
make MODE=debug all 
make[1]: Entering directory '/c/Users/Scott/Desktop/omnetpp-5.6.2/tempupdated/flora/src'
MSGC: LoRa/LoRaMacControlInfo.msg
LoRa/LoRaMacControlInfo.msg:20: Error: syntax error, unexpected NAME, expecting $end
make[1]: *** [Makefile:153: LoRa/LoRaMacControlInfo_m.h] Error 1
make[1]: Leaving directory '/c/Users/Scott/Desktop/omnetpp-5.6.2/tempupdated/flora/src'
make: *** [Makefile:2: all] Error 2
"make MODE=debug all" terminated with exit code 2. Build might be incomplete.

17:43:15 Build Failed. 3 errors, 0 warnings. (took 2s.221ms)

smblackledge, 2021-02-20:
Line 20 is the first import statement.

`
import inet.common.INETDefs;
import inet.common.Units;
import inet.linklayer.common.MacAddress;

namespace inet;

class LoRaMacControlInfo {
    MacAddress src;  // src DevAddr address (can be left empty when sending)
    MacAddress dest; // dest DevAddr address

    double LoRaTP;
    Hz LoRaCF;
    int LoRaSF;
    Hz LoRaBW;
    int LoRaCR;
    bool LoRaUseHeader;
}
`

mariuszslabicki, 2021-02-22:
Hi

As far as I remember on the website we have mentioned that the software doesn't run on Windows. Regarding the inet3to4 branch, it is under development, I plan to finish it rather soon and publish it, but right now it is 100% working.

Mariusz

smblackledge, 2021-02-23:
From flora.aalto.fi:

"For some reasons, FLoRa does not work on Windows, when compiled with clang."

The website only says that clang shouldn't be the compiler if Windows is where it will run. I'll make sure I'm not compiling with clang.

But if gcc is used, is flora expected to compile and run on Windows?

smblackledge, 2021-02-26:
Using the Windows version of the file. Changed CC=gcc, CXX=g++, and PREFER_CLANG=no in configure.user

Since you mentioned versions in your chat, I'm creating a flora environment exactly like the flora docs say.  This means, omnet++ 5.2.1, inet 3.6.3, and the currently submitted flora master.
 
Now, I have the correct omnetpp configured and made :-) with gcc.
 
Looking in the samples folder, which is what I am using for the Workbench. (Is it worth creating a "workbench" folder instead of using "samples"?)
 
Unlike the 5.6.2 samples folder, this one does *not* have an "inet" present.  In fact, inet 3.6.3 doesn't seem to exist.  Here are several of the "older releases" from the inet download page:
 

    INET 3.6.8 for OMNeT++ 5.3 (What’s New)
    INET 3.5.0 for OMNeT++ 5.1 (What’s New)
 ...
Notice that 3.6.3 isn't in the list.  What should I use in its place?

mariuszslabicki, 2021-04-10:
Updated to the lates OMNeT and INET.

closed 2021-04-10 by mariuszslabicki