florasim/flora #37: Get ID of the Node/GW

aruznieto, 2021-04-13:
Hi, I´m doing signals to compute the mean power received on a GW from each node, and I would like to know if it is possible to obtain the ID that Omnet++ assign to each Node/GW

mariuszslabicki, 2021-04-13:
I am not sure if I get on which level the information is expected. Have you checked component ID?
https://doc.omnetpp.org/omnetpp/manual/#sec:simple-modules:module-ids

aruznieto, 2021-04-13:
I am trying transmitter->getId() but this method does not return the Id that omnet++ asigns

mariuszslabicki, 2021-04-13:
From the top of my head I think that you need to catch a pointer of loranode object (not just transmitter), and then check nodeId. But do you need LoRaNode Id (as they are numbered in the node's vector), or just some unique number of an object?

aruznieto, 2021-04-13:
I need the unique number of the node, but as I can see, it isn´t necesary, with transmitter->getId() and receiver->getId() I obtain an unique ID to identify each signal. If you want to implement this on FloRa, I can send you the code in other moment

mariuszslabicki, 2021-04-13:
Hi, good to hear that. I think it is not needed in the common code.

closed 2021-04-13 by mariuszslabicki