florasim/flora #38: can't build Flora

abdelrahman960, 2021-04-25:
Hello, 
I need help with this error

make MODE=debug all 
make[1]: Entering directory '/c/Downloads/omnetpp-6.0pre10-src-windows/omnetpp-6.0pre10/samples/flora/src'
MSGC: LoRa/LoRaMacControlInfo.msg
LoRa/LoRaMacControlInfo.msg:20: Error: cannot resolve import 'inet.common.INETDefs'
make[1]: Leaving directory '/c/Downloads/omnetpp-6.0pre10-src-windows/omnetpp-6.0pre10/samples/flora/src'
LoRa/LoRaMacControlInfo.msg:21: Error: cannot resolve import 'inet.common.Units'
LoRa/LoRaMacControlInfo.msg:22: Error: cannot resolve import 'inet.linklayer.common.MacAddress'
LoRa/LoRaMacControlInfo.msg:31: Error: unknown type 'inet::MacAddress' for field 'src' in 'LoRaMacControlInfo'
LoRa/LoRaMacControlInfo.msg:32: Error: unknown type 'inet::MacAddress' for field 'dest' in 'LoRaMacControlInfo'
LoRa/LoRaMacControlInfo.msg:35: Error: unknown type 'inet::Hz' for field 'LoRaCF' in 'LoRaMacControlInfo'
LoRa/LoRaMacControlInfo.msg:37: Error: unknown type 'inet::Hz' for field 'LoRaBW' in 'LoRaMacControlInfo'
opp_msgtool: error in 1 file(s)
make[1]: *** [Makefile:155: LoRa/LoRaMacControlInfo_m.h] Error 1
make: *** [Makefile:5: all] Error 2
"make MODE=debug all" terminated with exit code 2. Build might be incomplete.

mariuszslabicki, 2021-04-25:
Have you added inet4.3 as a reference to FLoRa?

abdelrahman960, 2021-04-25:
> Have you added inet4.3 as a reference to FLoRa?

Yes. I did

abdelrahman960, 2021-04-25:
but now I have a different error :

make MODE=release all 
make[1]: Entering directory '/c/Downloads/omnetpp-6.0pre10/samples/flora/src'
LoRa/LoRaGWMac.cc
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:19:10: fatal error: inet/common/INETDefs.h: No such file or directory
   19 | #include "inet/common/INETDefs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: Leaving directory '/c/Downloads/omnetpp-6.0pre10/samples/flora/src'
make[1]: *** [Makefile:151: ../out/gcc-release/src/LoRa/LoRaGWMac.o] Error 1
make: *** [Makefile:5: all] Error 2
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

23:39:51 Build Failed. 3 errors, 0 warnings. (took 4s.298ms)

apd1999, 2021-05-01:
I dont suppose you've fixed this. Just got the exact same issue. 

make MODE=release all 
make[1]: Entering directory '/c/Users/alex/omnet_ws/flora/src'
LoRa/LoRaGWMac.cc
In file included from LoRa/LoRaGWMac.cc:16:
LoRa/LoRaGWMac.h:19:10: fatal error: inet/common/INETDefs.h: No such file or directory
   19 | #include "inet/common/INETDefs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:151: ../out/gcc-release/src/LoRa/LoRaGWMac.o] Error 1
make[1]: Leaving directory '/c/Users/alex/omnet_ws/flora/src'
make: *** [Makefile:5: all] Error 2
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

13:08:36 Build Failed. 3 errors, 0 warnings. (took 578ms)

abdelrahman960, 2021-05-02:
When I try to change something like version or settings other error comes out 
latest one is :

>>> defined at ../out/clang-release/src/LoRa/LoRaGWMac.o
>>> defined at libINET.dll

lld-link: error: duplicate symbol: inet::OperationalMixin<omnetpp::cSimpleModule>::isUp() const
>>> defined at ../out/clang-release/src/LoRa/LoRaGWMac.o
>>> defined at libINET.dll

lld-link: error: duplicate symbol: inet::OperationalMixin<omnetpp::cSimpleModule>::isDown() const
>>> defined at ../out/clang-release/src/LoRa/LoRaGWMac.o
>>> defined at libINET.dll
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:144: ../out/clang-release/src/libflora.dll] Error 1
make[1]: Leaving directory '/c/Users/aabde/Desktop/Last year/GP/simulation package/flora/src'
make: *** [Makefile:5: all] Error 2
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

10:07:35 Build Failed. 2 errors, 0 warnings. (took 18s.914ms)

mariuszslabicki, 2021-05-02:
I am not sure from this error msg, have you been able to compile inet?

husnazabidin, 2021-06-10:
Hi,

I installed omnetpp 6.0 pre 11 with inet-4.3.2 on ubuntu 20.04 LTS. I made changes in configure.user as suggested by apd1999 in #39 . I have no problem in building the inet-4.3.2 but I still have error when building the flora. It says that

/src/LoRaPhy/LoRaAnalogModel.cc:16:10: fatal error: LoRaPhy/LoRaAnalogModel.h: No such file or directory

Can anyone help me with this? Thank you.

abdelrahman960, 2021-07-22:
doesn't work on windows so far but did on ubuntu 20

closed 2021-07-22 by abdelrahman960