florasim/flora #42: flora end to end communication

salahmoh2001, 2021-07-06:
if any one help me to stop broadcast between Nodes ,  my case All LoraNode to send to LoraGW   At the same time

mariuszslabicki, 2021-07-06:
Hi! I am not sure if I understand the issue, the current implementation of LoRaApp covers sending packets after random times, it is defined as:
`**.timeToFirstPacket = exponential(1000s)`
`**.timeToNextPacket = exponential(1000s)`
https://github.com/mariuszslabicki/flora/blob/ac94052b57d45a7680700d7ff23b7f9601cfe195/simulations/omnetpp.ini#L31

Those parameters can be set individually for each node, so how is that possible that all nodes send data at the same time?

salahmoh2001, 2021-07-06:
from which file.cc i can modify this issue and stop broradcast

mariuszslabicki, 2021-07-06:
Oh, you want to have device do device communication. You need to reimplement at least MAC layer.

salahmoh2001, 2021-07-06:
from where exact i can modify this issue, Thanks A lot

salahmoh2001, 2021-07-06:
Oh, you want to have device do GW communication ???

salahmoh2001, 2021-07-06:
please can give me more Detalis about 👍 
- send message brodcast: Loraphy\LoRaTransmitter.cc    or different location 
- from where i can change (parameters can be set individually for each node, so how is that possible that all nodes send data at the same time)
thanks

mariuszslabicki, 2021-07-06:
I think you can start with files covering LoRa MAC layer (LoRaMAC.**).

I don't understand the issue that all nodes send data at the same time, definitely they are not sending data at the same time if you don't enforce them to do that.

salahmoh2001, 2021-07-06:
please how i can stop Broadcast and using unicast sending !!!!

mariuszslabicki, 2021-07-06:
This is not implemented. If you want to do that you need to implement it in files that I mentioned in previous messages.

closed 2021-07-06 by mariuszslabicki