florasim/flora #45: incomplete build issue

abdelrahman960, 2021-07-15:
Hello everybody

I'm facing some kind of errors I couldn't find any solution for
those are :
lld-link: error: too many errors emitted, stopping now (use /errorlimit:0 to see all errors)
collect2.exe: error: ld returned 1 exit status
make[1]: *** [Makefile:144: ../out/gcc-debug/src/libflora_dbg.dll] Error 1
make[1]: Leaving directory '/c/Users/aabde/Downloads/omnetpp-6.0pre11/samples/flora/src'
make: *** [Makefile:5: all] Error 2
"make MODE=debug all" terminated with exit code 2. Build might be incomplete.

felipealencar, 2021-07-16:
> Hello everybody
>
> I'm facing some kind of errors I couldn't find any solution for
> those are :
> lld-link: error: too many errors emitted, stopping now (use /errorlimit:0 to see all errors) [...]

Hey, Abdel!
You seem to be using Windows as environment. I confess I failed trying over and over again building FLoRa in Windows. My solution was creating a Ubuntu 20 VM and building FLoRa there.

You could try using clang or there compiler. Also try to check you OMNeT and INET versions.

abdelrahman960, 2021-07-21:
it didn't work on windows,but did on Ubuntu 20.04

closed 2021-07-21 by abdelrahman960