florasim/flora #52: numCollision recording grey-out

lingcy0328, 2021-09-20:
Hi, i'm confusing on why the numCollision statistic recording is commented out. Did it means that the collision is not considered anymore in the LoRaMac.cc ？Or did it moved to other .cc file?

mariuszslabicki, 2021-09-20:
Hi

If I remember correct I have commented it out as I have not verified if number of collisions has been counted accurately.

lingcy0328, 2021-09-21:
Hi, I found another numCollisions statistic recording in LoRaReceiver.cc, that is for physical layer. Perhaps you replace the one in LoRaMac.cc with that? Also, I found that the numCollision in LoRaMac.cc is not used throughout the .cc file but only initialized in the initialization().