florasim/flora #54: Issues building Flora on Ubuntu 20

NicolasChollet51, 2021-11-23:
Hi everyone,

I'm having an issue building the project in Omnet 6,  I did Make sure that the FLoRa project has INET project (inet4.3) as a project reference (Project > Properties > Project References) as you can see in the first  picture. The second pictures show the line of error at LoRa/LoRaMacControlInfo.msg:20 .

Inet is building ok, I installed it via the omnet functionallity (at first install of omnet) , do you think it can come from there

My hypothesis is that the link between inet and flora did not happen but how can i fix it? The error tell me that is a synthax one but i gess that it's more about the import inet. Do you have any advise?

Thanks a lot!

15:45:11 **** Incremental Build of configuration release for project flora ****
make MODE=release all 
make[1]: Entering directory '/home/nico/omnetpp-5.7/samples/flora/src'
MSGC: LoRa/LoRaMacControlInfo.msg
LoRa/LoRaMacControlInfo.msg:20: Error: syntax error, unexpected NAME, expecting $end
make[1]: Leaving directory '/home/nico/omnetpp-5.7/samples/flora/src'
make[1]: *** [Makefile:157: LoRa/LoRaMacControlInfo_m.h] Error 1
make: *** [Makefile:5: all] Error 2
"make MODE=release all" terminated with exit code 2. Build might be incomplete.

15:45:11 Build Failed. 3 errors, 0 warnings. (took 175ms)

[image]

[image]

mariuszslabicki, 2021-11-23:
Hi Nicolas,

Recently many dependencies are in rapid update process, and proper dependencies versions are crucial. I plan to release stable version in few weeks, where probably all will be documented, but I am waiting before omnet and inet are released in stable versions.

Before that you can use the following configuration suggested by FLoRa user:
https://github.com/mariuszslabicki/flora/issues/48#issuecomment-952662118

NicolasChollet51, 2021-11-24:
Hello, well thanks a lot i did succeed the build with this specs :) sorry for the bother and thanks for the project!

closed 2021-11-24 by NicolasChollet51

Precious1101, 2022-04-29:
> Hi Nicolas,
> 
> Recently many dependencies are in rapid update process, and proper dependencies versions are crucial. I plan to release stable version in few weeks, where probably all will be documented, but I am waiting before omnet and inet are released in stable versions.
> 
> Before that you can use the following configuration suggested by FLoRa user: [#48 (comment)](https://github.com/mariuszslabicki/flora/issues/48#issuecomment-952662118)
Hi Mariusz,

Since the stable version OMNet++6.0 is released, could you please update stable version too?
I am new to OMNet++ and I use Windows10.
Before the update I would have to switch to Ubuntu, another stranger.

Great thanks!

Precious1101, 2022-04-29:
> Hi Nicolas,
>
> Recently many dependencies are in rapid update process, and proper dependencies versions are crucial. I plan to release stable version in few weeks, where probably all will be [...]

Hi Mariusz,

Since the stable version OMNet++6.0 is released, could you please update stable version too?
I am new to OMNet++ and I use Windows10.
Before the update I would have to switch to Ubuntu, another stranger.

Great thanks!

Precious1101, 2022-04-29:
> Hi Nicolas,
>
> Recently many dependencies are in rapid update process, and proper dependencies versions are crucial. I plan to release stable version in few weeks, where probably all will be [...]

Hi Mariusz,

Since the stable version OMNet++6.0 is released, could you please update stable version too?
I am new to OMNet++ and I use Windows10.
Before the update I would have to follow #60  and switch to Ubuntu, another stranger for me.

Great thanks!