florasim/flora #57: ADR doesn't seem to work on the network server

Jung-HyeonHO, 2022-02-13:
Hello 

I'm running an ADR simulation of FLoRa  

I setting **.networkServer.**.evaluateADRinServer = true

When I start all nodes with SF12 but there was no change

Do I need to make any other settings?  

would you help me?

Jung-HyeonHO, 2022-02-14:
My guess is that the lora node does not receive the SF calculated by the network server.

mariuszslabicki, 2022-02-14:
Hi

It is a bit surprising, before we get to the debugging of ADR - have you verified that messages from nodes are reached by GW, and if NetworkServer is creating ADR packages, are they received by end nodes?

camejuanm, 2023-09-01:
Hi, I encountered the same issue. 
[image]

as you can see from the image, the end nodes does receive ADR commands

camejuanm, 2023-09-01:
hi @Jung-HyeonHO do you still have the same issue? have you found what is causing Network Server ADR not affecting end node SF/TP?