florasim/flora #64: simulation can't complete

Sehamibrahem, 2022-06-16:
hi 
I'm new in flora, when I made build to the file, the building process complete and the qenv screen appear and I can see the packets transmission but the simulation process remain at (0%) and the simulation files don't appear. I don't know the reason.
can any one help me?
I'm working on windows 10, omnetpp 5.6 pro 12, inetmannet4.x
thank you
[image]

mariuszslabicki, 2022-06-20:
Hi!

We have published a new version of FLoRa, and added also manual how to setup a basic simulation. You can find more details here:
https://flora.aalto.fi/

closed 2022-06-20 by mariuszslabicki

Sehamibrahem, 2022-06-20:
Hi

HI Mariusz Slabicki

Thank you for your attention. I have a question about your code for
calculating transmission time depending on duty cycle (new feature ) and
coding rate. Can you explain how you

calculate the transmission time in (simple LoRaAPP.CC) file. as for example
at sf = 7, you used the equation (49.408 + loRaCR*7.168) / loRaDC / 10.0.
what's these constants???  I can't understand it.
thank you for your help
seham ibrahem
PhD student

‫في الاثنين، 20 يونيو 2022 في 10:16 ص تمت كتابة ما يلي بواسطة ‪Mariusz
Slabicki‬‏ ***@***.***‬‏>:‬